AislerHQ/PushForKiCad #1: Python exception on OSX - dark mode?

tristancollins, 2022-01-06:
I get the attached error when loading the PCB editor.

Kicad 6.0 on latest mac OSX

"AttributeError: 'PishForKiCadPlugin' object has no attribute 'dark_icon_file_name'"

Perhaps related to dark mode?
[image]

The icon appears as a generic extension icon along the toolbar.

tristancollins, 2022-01-06:
The search result from below indicate this may be a KiCAD issue?

https://gitlab.com/search?group_id=6593371&scope=blobs&search=dark_icon_file_name

```
If icon is not specified a generic tool icon will be used.

An additional property `self.dark_icon_file_name` may be specified to provide an alternate icon for
dark window themes.  If this property is not present, the icon specified in `icon_file_name` will
be used when the dark theme is active;

```

PatrickFranken, 2022-01-07:
Thanks for the report including the actual solution to this :) Will be included in the next release.

commit 657f0560 (2022-01-07): Fix #1
closed 2022-01-07 by PatrickFranken