AislerHQ/PushForKiCad #7: KiCad 6.99 crash - unsupported operand type(s) for /: 'SwigPyObject' and 'float'

pointhi, 2022-01-28:
Using this plugin with the current nightly crashes the plugin due to changes how the orientation is stored.

```
wig/python detected a memory leak of type 'EDA_ANGLE *', no destructor found.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/thomas/.local/share/kicad/6.99/3rdparty/plugins/com_github_aislerhq_PushForKiCad/push_thread.py", line 114, in run
    'rotation': f.GetOrientation() / 10.0,
TypeError: unsupported operand type(s) for /: 'SwigPyObject' and 'float'
```

The fix is to use the following code instead in current KiCad versions:

```python
'rotation': f.GetOrientation().AsDegrees(),
```

PatrickFranken, 2022-02-02:
Do you have any clue if .AsDegress() is also available in v6 stable release?

pointhi, 2022-02-04:
No, AsDegree was introduced with the new angle type. I suggest to check if orientation is an integer I think, and otherwise use the new code.

smurfix, 2022-03-05:
The new angle code is unlikely to go away, so my patch just fetches the angle in the "new" way and falls back to assuming integers-or-whatever if that causes an exception.

PatrickFranken, 2022-03-06:
Solved by https://github.com/AislerHQ/PushForKiCad/pull/13

closed 2022-03-06 by PatrickFranken