AislerHQ/PushForKiCad #8: Unable to push project to Aisler, if project has been deleted

PeterMarkl, 2022-01-30:
As the title says. I have uploaded a project to Aisler to test the new plugin, then deleted this project online.
One day later I changed some things in Kicad and wanted to upload it to Aisler and got the Error: "Already logged in" ..

That wasn't very useful, but I figured out that I had to delete "AISLER Project ID: xxxxxxxxxx" in Comment 4 in Kicad Leiterplattendesigner (i guess pcb-designer) -> Datei (File) -> Seite einrichten (Setup Page?) - Sorry I run the german version of Kicad.

After that, I could again upload my project, because I guess it got a new ID from Aisler.

PatrickFranken, 2022-02-02:
That is exactly how it works, we use the AISLER Project ID to identify the project on our end and thus add a new revision instead of adding a new project each time. However, the behavior you described is not intended. Do you have any clue how we should handle this?

PeterMarkl, 2022-02-02:
I think if the user gets a more informative Error message on your website, then it is no problem. 
At the moment it really just says something like "You are already logged in", which has nothing to do with the error itself. Tested on Chrome - I don't know if its the same on Firefox etc.

Additionally you should add some information, where the AISLER Project ID is to find in Kicad and that the problem is resolved, if the user deletes the ID.

suuppl, 2024-04-21:
I had the same problem. As a workaround, you can force the plugin to create a new project by deleting the line `AISLER Project ID: XXXXXXXX` from Comment4 in the PCB Title Block.

tbolender, 2026-09-11:
Close as error message is displayed in this case.

closed 2026-09-11 by tbolender