AislerHQ/PushForKiCad #10: Upload stuck

zonque, 2022-02-03:
The upload progress remains stuck:

[image]

Application: KiCad PCB Editor

Version: 6.99.0-1.20220202gitc8d14ad.fc35, release build

Libraries:
	wxWidgets 3.0.5
	FreeType 2.11.0
	libcurl/7.79.1 OpenSSL/1.1.1l-fips zlib/1.2.11 brotli/1.0.9 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.46.0 OpenLDAP/2.4.59

Platform: Linux 5.15.16-200.fc35.x86_64 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Feb  2 2022 18:17:25
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.79.1
	ngspice: 36
	Compiler: GCC 11.2.1 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

pointhi, 2022-02-04:
That issue is caused by #7

zonque, 2022-02-04:
> That issue is caused by #7

I see. Closing hence. Thanks!

closed 2022-02-04 by zonque