AislerHQ/PushForKiCad #12: Push to Aisler button?

Bucknalla, 2022-02-28:
This isn't an issue (Aisler is awesome and so is your KiCAD plugin!) but I was thinking about how it might make sense to have something like a 'send to Aisler' button that people could embed on their GitHub projects that could easily wrap the source of the project (or maybe even the release artifacts) and push them directly to Aisler, in order to allow people to place order for the PCBs in that repo?

This idea was inspired by our own 'deploy with balena' [button](https://www.balena.io/docs/learn/deploy/deploy-with-balena-button/) which you can use to deploy software with. 
For example:

[image]

It's not always feasible to build and ship hardware at small volumes (especially in 2022!) but this way it would be one click for a user to order an open source PCB design, through Aisler.

Thanks for your great work, big fan of Aisler and how awesome/easy it is to prototype with! 

P.S. let me know if this is the wrong place for this feedback!

tbolender, 2022-02-28:
Hi @Bucknalla, thank you very much for the input! We are always eager to get feedback from you. We therefore started our [community](https://aisler.community/) with the possibility to share ideas like that. Would you mind creating a topic there as well? It's also a perfect place to get feedback from others.

Bucknalla, 2022-03-01:
Will do @tbolender !