AislerHQ/PushForKiCad #14: Fails to Upload

SilvanRehm, 2022-04-26:
When trying to upload a project (attached), the following happens:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/home/silvan/.local/share/kicad/6.99/3rdparty/
plugins/com_github_aislerhq_PushForKiCad/push_thread.py", line 50, in run
    popt.SetExcludeEdgeLayer(True)
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer'
```

[Cosmic_Calibration_PCB.zip](https://github.com/AislerHQ/PushForKiCad/files/8560925/Cosmic_Calibration_PCB.zip)

KiCAD Version: (6.99.0-1615-g403992a6f4-dirty), release build
Plugin Version: 0.14

s-light, 2022-05-01:
i just encountered the same error:
first reported it in the [community-forum](https://aisler.community/t/kicad/88/16?u=s-light)
and then found this report. ( i will link there to here..)

## steps to reproduce
- start kicad from the terminal
- create empty project
- open PCB-Editor
- just define a edge-cut line
- click on the aisler upload-button
- watch the terminal:
```bash
$ kicad-nightly 
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/home/stefan/.local/share/kicad/6.99/3rdparty/plugins/com_github_aislerhq_PushForKiCad/push_thread.py", line 50, in run
    popt.SetExcludeEdgeLayer(True)
  File "/usr/lib/kicad-nightly/lib/python3/dist-packages/pcbnew.py", line 8900, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, PCB_PLOT_PARAMS, name)
  File "/usr/lib/kicad-nightly/lib/python3/dist-packages/pcbnew.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer'
```
so the upload fails / does not really start...

my version
```
Application: KiCad

Version: 6.99.0-unknown-484fabf336~148~ubuntu21.10.1, release build

Libraries:
	wxWidgets 3.0.5
	FreeType 2.10.4
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.74.0 OpenSSL/1.1.1l zlib/1.2.11 brotli/1.0.9 libidn2/2.3.1 libpsl/0.21.0 (+libidn2/2.3.0) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3

Platform: Linux 5.13.0-41-generic x86_64, 64 bit, Little endian, wxGTK, KDE, x11

Build Info:
	Date: May  1 2022 08:10:43
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.74.0
	ngspice: 35
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

PatrickFranken, 2022-05-03:
This is super strange as actually https://gitlab.com/kicad/code/kicad/-/blob/6.99.0/pcbnew/pcb_plot_params.h#L132 hasn't been touched lately. I'll consult the KiCad folks on this, keep you posted.

PatrickFranken, 2022-05-10:
[PushForKiCad-0.1.5-pcm.zip](https://github.com/AislerHQ/PushForKiCad/files/8661933/PushForKiCad-0.1.5-pcm.zip)
Here is an updated version including a fix (https://github.com/AislerHQ/PushForKiCad/commit/d3c9ab82f5d34d3e3027e382d4656783f73d6055) to solve the issue. Could one of you @s-light  @SilvanRehm please validate?

s-light, 2022-05-10:
i hope i find some time tomorrow evening!
thanks for investigating!

s-light, 2022-05-23:
sorry for the delay!
i just checked the updated version - and it is working nicely!
@PatrickFranken  thanks for fixing it!

zonque, 2022-07-07:
This fix works for me as well. Any plans o making a release with this included?

closed 2026-09-11 by tbolender