AislerHQ/PushForKiCad #21: plugin does not respect "exclude from bom"

Fuzzlix, 2023-02-28:
some parts are set as "exclude from bom" in eeschema but are included in the uploaded part list. This includes in my case:
- manual placed footprints like Silkscreen graphics (set as exclude from BOM)
- TC2050 connector (set as exclude from BOM)
- a footprint with the board outline (set as exclude from BOM)
- ...

PatrickFranken, 2023-03-14:
We actually  make use of the Not in BOM flag, see https://github.com/AislerHQ/PushForKiCad/blob/main/src/push_thread.py#L113
However we we still import the parts but mark them as do not place during import.

closed 2023-03-14 by PatrickFranken

PatrickFranken, 2023-03-14:
Just doubled checked, if Not in BOM is used it will show up as "exccluded" on the BOM after import. Could you send a mail to hello@aisler.net with the file attache? Or if it is not confidential, just attach it right here.