AislerHQ/PushForKiCad #22: PCB export issue

msasdk, 2023-03-09:
Hello!
I have created this PCB in KiCad 7.0:
[problematic.zip](https://github.com/AislerHQ/PushForKiCad/files/10929157/problematic.zip)
[image]
[image]

When exported as gerber, things looks as they are intended:
[image]
However, when exported using the Aisler plugin, or if the kicad_pcb fileis imported directly on Aisler.net, I get following result.

 [image]

PatrickFranken, 2023-03-14:
Hi! I guess this issue is more KiCad related. If I open the generated gerber files using the Ucamco reference viewer I get a list of issues. These issues also cause our CAM to misbehave.
[image]

msasdk, 2023-03-14:
But as stated in the issue description, this does not occur if I generate the gerber with KiCad without the plugin.

msasdk, 2023-03-14:
These gerbers are generated directly from KiCad
[grb.zip](https://github.com/AislerHQ/PushForKiCad/files/10969840/grb.zip)

PatrickFranken, 2023-03-14:
You are right on this! However I have no clue what causes KiCad to plot these corrupted Gerbers as the plugin actually uses the same interface as the UI. I'll check with the KiCad developers!

Golany2806, 2023-04-11:
Seems there is a problem using push to aisler using kicad...it connecting to microsoft edge without loading the files...