AislerHQ/PushForKiCad #30: Bug: Vias always tented with V8.0.8

ModuloFS, 2025-01-13:
As with KiCad Version 8.0.8 the vias are always covered with solder mask in the exported gerber data, no matter what settings are done in the board proberties. 

With version 8.0.2 everything worked fine. 

Does anybode have the same issue?

ModuloFS, 2025-01-13:
Problem seems to be gone at least in the version: 9.0.0-rc1-459-gb92bef8eee

PatrickFranken, 2025-01-21:
A few settings are set by us, no matter which settings are used on the project itself. See
https://github.com/AislerHQ/PushForKiCad/blob/main/src/push_thread.py#L35
However, via covering isn't any of these options. Do you know if the default in 8.0.2 might have changed? And would it help to set a sane default by the plugin?

ModuloFS, 2025-01-21:
What do you mean with "default"?

I think it is special related to the KiCad Version 8.0.8 because now i downgraded temporarily to 8.0.7 and everything is ok. As well as i said in the previous post it seems also working fine in the new version 9.

So it tends to be just KiCad related and hopefully gone with the new release of V9 in the near future.

ModuloFS, 2025-05-23:
Issue solved in KiCad 9

closed 2025-05-23 by ModuloFS