AislerHQ/PushForKiCad #31: Feature: Give Aisler-order same name as project/PCB file

Karim-at-Powl, 2025-02-04:
Hi, awesome awesome add-on that makes Aisler my No. 1 PCB Fab choice now! But, while awaiting the first 3 orders, I realised that the order is named "[Sandbox/tmp2b__XXXX" instead of taking the KiCad Project Name / PCB file name as a reference. 
Because when scrolling through my accounts orders, I have no clue at all which PCB is which. Can you add this feature? Thanks!

Otherwise works great!

tbolender, 2025-02-07:
Thanks for the kind words! We normally take the title of the board, I guess you did not specify it? I'll check whether we can use name of the PCB file as fallback.

Karim-at-Powl, 2025-02-07:
Oh that makes so much sense! Thanks a lot & good to know and for the future.

closed 2025-02-07 by Karim-at-Powl